nodestark/mdb-esp32-cashless #1: BOM part list

veyselozdemir29, 2024-07-12:
Hello, First of all, I would like to thank you very much for sharing this project. Can you share the parts list used? There are only position numbers in the kicad file, many component information appears to be missing. can you help me?

nodestark, 2024-07-12:
Hi, I'll put that info on Kicad:

You can use a 240ohm and 390ohm for adjustable regulator (to reach 3v3);
For LC circuit (WiFi antenna), uses 1.8pf capacitor and 2.4nh inductor;
22pf capacitor to crystal;

The other resistors, you can use 1k;

The other capacitors is not necessary (only if the power is not stable)

lynxis, 2025-02-10:
I looked into the client/slave schematics. The crystal should be a 40 MHz from reading the esp32-s3 recommendation, it is marked as 40kHz.

nodestark, 2025-05-08:
You are correct, it's 40 MHz.

closed 2025-05-08 by nodestark