nodestark/mdb-esp32-cashless #2: Trying to print the PCBs

maleksafadi, 2024-09-05:
Thank you for sharing this valuable project.
Actually I'm not familiar with electronics but I can do coding for embedded devices like yours.
 I tried to print the PCBs from pcbway.com or jlcpcb.com but unfortunately both giving issues because there is missing components.
Could you please share the gerber and bom files for each pcb? also the usb type c and mdb connectors not added to the design.
If you can make them as per the pictures you provide in the repository that's will be appreciated.

So I can print them directly.

maleksafadi, 2024-09-24:
Hi,
Can you add the the production files?

nodestark, 2025-05-08:
Hi friend!
It's on PCBway!!!

https://www.pcbway.com/project/shareproject/mdb_esp32_cashless_bc6bf8d8.html

closed 2025-05-08 by nodestark