nodestark/mdb-esp32-cashless #3: Any examples / getting starting guide

AlexGx, 2024-09-12:
Your project looks promising. Can you provide any example applications, for example cashless device emulation.

Thanks.

nodestark, 2024-09-12:
In that project are slave and master, the master emulates the machine.

The slave is cashless, the project has both.

tjunussov, 2025-05-03:
can't flash master via IDF, i tried press button but it says 

esptool.py v4.8.1
Serial port /dev/tty.usbmodem101
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: cc:ba:97:27:1a:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
A fatal error occurred: No serial data received.
 *  The terminal process

tjunussov, 2025-05-07:
I tried this, also no luck, i press reset before plugin to USB, and in monitor mode i got waiting for download

python esptool.py -p /dev/tty.usbmodem101 -b 115200 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode qio --flash_freq 40m --flash_size 8MB 0x0 bootloader/bootloader.bin 0x10000 mdb-master-esp32s3.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.8.1
Serial port /dev/tty.usbmodem101
Connecting...
Chip is ESP32-S3 (QFN56) (revision v0.2)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: cc:ba:97:27:1a:38
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).

nodestark, 2025-05-08:
Hi friend,
That sdkconfig was for another board, sorry about that;
Please, update your branch and try again.

idf.py flash monitor

Regards.