nodestark/mdb-esp32-cashless #7: Schematic issue

azamat5991, 2025-03-07:
Hi,
I've checked the schematics and found that you used 40kHz clock instead of 40MHz.
Does your board work?

nodestark, 2025-03-07:
Hi friend, sorry for that, you are correct, it's 40Mhz.

azamat5991, 2025-03-07:
have you tested your board? and does it work?

nodestark, 2025-03-07:
I'll receive that board next week, this had some noise problems, was fixed.
So only next week to do tests. Before was esp32 devboard only.

Regards,

Leonardo Soares

azamat5991, 2025-03-07:
ok, thanks.
Will wait for updates:)

maleksafadi, 2025-04-06:
You did a great job, and I think if you update it and fix the issues I have to donate for you due to this effort.
Please keep going and do the update asap.

tjunussov, 2025-04-23:
Hi! there, any news on this topic ? so we should go with 40Mhz only ?

nodestark, 2025-04-23:
Hi! Thanks for checking in. Yes, based on the current requirements and
constraints, we should proceed with *40 MHz only*. Let me know if you need
any help adjusting the configuration.

maleksafadi, 2025-04-24:
Hi @nodestark, I tried to print it in pcbway, but they said some of production files are missing, could you please make it final in order to be able to print it?

maleksafadi, 2025-04-26:
Thank you so much for your update.
I tried to download the files:
[mdb-slave-esp32s3.pdf](https://github.com/user-attachments/files/19922711/mdb-slave-esp32s3.pdf)
[mdb-slave-esp32s3-pcb.zip](https://github.com/user-attachments/files/19922712/mdb-slave-esp32s3-pcb.zip)
Both files giving (not found).
Please check.
and regarding pcbway, when I tried to print your project last month they reply to me as below:

**There are some vias be designed on the SMD pads, pls advise if proceed with via-in-pad process?
If yes, this service is chargeable (over $100usd) and vias are to be non-conductive epoxy filled, plated over with copper and planarized.
If not, in order to guarantee assembly quality, it's suggested you can update design, move vias away from SMD pads and attach the new gerber files by email.**

If you can fix it, I will be appreciated.

nodestark, 2025-04-26:
Hi friend,

Thank you for your feedback!

The project is now available on PCBWay at this link:

👉
https://www.pcbway.com/project/shareproject/mdb_esp32_cashless_bc6bf8d8.html

Please feel free to check it there.
Let me know if you have any other questions.

Best regards,

maleksafadi, 2025-04-26:
Thank you for quick reply, I added this project to pcbway cart and now it's under review.
I will update you based on thier feedback.
Really appreciated.

maleksafadi, 2025-04-28:
I received from Pcbway that the bom list not available, could you please review your project in pcbway and share the bom list?
thanks

nodestark, 2025-04-29:
Hi friend,
Ok, it's there!

https://www.pcbway.com/project/shareproject/mdb_esp32_cashless_bc6bf8d8.html

maleksafadi, 2025-06-30:
Hi, today I have recived 2 pcs of the pcb ordered from pcbway, when I tried to test the first one, I connect the mdb connector then the mcu (ESP32-S3 burned directly.
I tried the other one by connect the type-c usb to my laptop but it's giving  usb device not recognized frequantly.
I think you missed something. please see the attached.

[image]

maleksafadi, 2025-07-02:
Thanks @nodestark for your reply.
the burned pcb was connected to the MDB which is 24v because the vending machine MDB connectors provide 24v to 34v.
For the second board that giving (usb device not recognized), I tried to connect it directly to my laptop without power but unfortunately it's not working, I have attached a video showing what is happening when I connect it to usb.

https://github.com/user-attachments/assets/ffa7fc4a-df7e-4464-967b-a416f3c9d5d3

Your support is really appreciated.

nodestark, 2025-07-02:
Friend, try to connect USB with button pressed, to enter in boot mode. In
that moment the chip has no USB code to use the drive.

maleksafadi, 2025-07-03:
> Friend, try to connect USB with button pressed, to enter in boot mode. In
> that moment the chip has no USB code to use the drive.
> […](#)

I have tried as you mentioned and it seems working and the error (usb device not recognized) appears once, but the problem is the port not discovered by Arduino IDE, also when I go to the device manager it's listed as Unknown USB Device (Device Descriptor Request Faield), see the screenshots.

[image]

[image]

nodestark, 2025-07-04:
Hi friend, success?

maleksafadi, 2025-07-04:
> That MCU is the esp32-s3, it seems drive not installed; I have used Linux,
> and we can use the lsusb; Could you check the crystal (40Mhz), we can check
> the 3V regulator. If you feel fit, you can try charging it with external 3V
> power, on the left pins.
> […](#)

Dear @nodestark ,
Could you please share with me a screenshot about it, because I'm not familiar with circuit boards like you.
Also I have another question please, if we can make the usb working well and the mdb interface cannot work with 24v as the vending machine power supply, how I can solve this issue?

nodestark, 2025-07-05:
Hi @maleksafadi 
Yes, the next step is to assemble it in production mode — it's currently only in development.
It unlocks access to cashless payments, remote monitoring, and IoT integration, empowering developers and businesses to modernize machines with low-cost, fully customizable hardware and firmware.

By breaking the closed nature of the vending industry, it democratizes automation and fosters innovation.

I'd love to hear more about your plans. That project aims to reach as many developers as possible, right?

azamat5991, 2025-07-06:
Hi, 
Due to issues of the board, I bought esp32 s3 dev boards, how can I test it in master-slave configuration?

nodestark, 2025-07-07:
[image]
[image]
[image]
[image]

maleksafadi, 2025-07-07:
> [image] [image] [image] [image]

this is nice work, have you tested this board?

nodestark, 2025-07-09:
Hi @azamat5991 , yes! You can use[mdb-master-esp32s3](https://github.com/nodestark/mdb-esp32-cashless/tree/main/mdb-master-esp32s3)

nodestark, 2025-08-05:
[image]

nodestark, 2025-08-05:
> T-V17W08384A.jpg (view on web)
>
> Hi @nodestark <https://github.com/nodestark> ,
> Have you done with updating the pcb to work with 24v as power that comes
> from the vending machine?

Yes friend, it supports up to 27V, and RC circuit on enable pin. BUT it's
in PCBway yet, they send me pictures.

avillegas, 2025-08-06:
Hello, thank you for your project !! , I will try this board ASAP. 

Regards