nodestark/mdb-esp32-cashless #14: UFP Rd Pull-Down Resistors

woto, 2025-08-01:
Hi,
Perhaps you already saw the first issue I created. I'm continuing to research your schema. I'm also not sure if this is a real issue or not, but this is what I found in the specification:

[image]

but I see 2 resistiors with 3.3K on the schema

[image]

leonardobsi, 2025-08-02:
Hi friend, kicad project updated, mdb-slave-esp32 updated; Soon I'll update mdb-master-esp32 too, to support high input voltage

leonardobsi, 2025-08-02:
@woto

leonardobsi, 2025-08-02:
> Hello everyone,
> thank you for working on this project. I'm also currently looking into
> options to upgrade our beverage and snack vending machines with a modern
> payment and inventory management system. I'm curious to see how things
> develop here. Best of luck with the project!
>
> By the way, the PCBWay link in the README doesn't seem to be working:
>
> https://www.pcbway.com/project/shareproject/mdb_esp32_cashless_bc6bf8d8.html

Hi, thank you a lot! All are welcome, the pcbway is out for while, I'm
waiting the last version ready, it's there. Soon we'll have more stacks.

Regards,

closed 2025-08-04 by woto